eclipse-openbsw/openbsw #9: UNCOPYABLE vs. uncopyable

rolandreichweinbmw, 2024-10-29:
As I understand, the Macro UNCOPYABLE is included in estd's uncopyable.h for historical reasons?

Instead, the base class estd::uncopyable can be used instead consequently, since using C++ means is mostly better than preprocessor means.

vladyslavmarkovaccenture, 2024-11-07:
> As I understand, the Macro UNCOPYABLE is included in estd's uncopyable.h for historical reasons?

Exactly. The `UNCOPYABLE` macro will be removed in the future as it is deprecated.

> Instead, the base class estd::uncopyable can be used instead consequently, since using C++ means is mostly better than preprocessor means.

The current preferred way to make a class non-copyable is to declare the copy constructor and copy assignment operator as `= delete`d under the `public:` access specifier (see [C++ core guidelines: C.81](https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#c81-use-delete-when-you-want-to-disable-default-behavior-without-wanting-an-alternative)).

A less preferred, more Boost-like approach is private inheritance from `estd::uncopyable`, as it:
- Clutters the class hierarchy
- Creates inconsistency in class interface definitions when applying the [Rule of Five](https://en.cppreference.com/w/cpp/language/rule_of_three)

rolandreichweinbmw, 2025-02-21:
This issue can be closed when the ETL PR is done.

closed 2025-07-10 by mthiede-acn2