eclipse-openbsw/openbsw #32: release estd as an own repo/project?

questor, 2024-12-04:
Hi! thanks for releasing this, it's really interesting. By looking at the source code I've found that you "indirectly" released the sourcecode of estd as part of this repo, do you plan to release estd also as an extra project (potentially in the esrlabs-orga as own repo)? There was a preview-release of only the vector, but sadly up to now no complete release of the library.

ThoFrank, 2024-12-05:
Hi there!
I am not aware of any (internal) discussions about that yet. Thanks for bringing it up! I think it's easier for us to keep everything in just one repo.

But maybe a [josh](https://github.com/josh-project/josh) proxy can help you extract estd from openbsw ([example](https://josh-project.github.io/josh/guide/gettingstarted.html#cloning-a-part-of-the-repo)).
Also this way it would still be easy for you to contribute back any changes.

questor, 2024-12-05:
I already came across josh, but up to now had no real usecase for it... thanks!

closed 2024-12-05 by questor