eclipse-openbsw/openbsw #37: ### OpenBSW Support for MPC5746G (PowerPC, Multicore) and AUTOSAR MCAL

josephsherif, 2024-12-17:
We are currently working on a project using the NXP MPC5746G, which:
- Is multicore and based on the PowerPC architecture.
- Utilizes AUTOSAR MCAL drivers provided by NXP (written in C).

We would like to know:

1. Does OpenBSW support MPC5746G or the PowerPC architecture in general?
2. Does OpenBSW have any built-in support for multicore microcontrollers?
3. Since our MCAL drivers are written in C and OpenBSW is written in C++, will there be any compatibility issues?

Any guidance or recommendations for integrating OpenBSW with our hardware and drivers would be appreciated!

poula10, 2024-12-18:
"I’m currently facing the same challenges with the NXP MPC5748G. Specifically, I’m also trying to determine whether OpenBSW supports the PowerPC architecture and how to handle the integration with C-based MCAL drivers. If you find any solutions or receive helpful recommendations, I’d greatly appreciate it if you could share them. Likewise, I’ll share any updates I come across. Let’s collaborate to tackle this!"

JohnathanGamal, 2024-12-18:
I’m also working with the MPC5748G and trying to understand how well OpenBSW supports the PowerPC architecture and multicore setups. The integration with NXP’s C-based MCAL drivers is another puzzle I’m looking to solve.

It would be nice if one of the contributors could provide some insights or guidance on this. OpenBSW seems promising, and I’d love to know how we can best adapt it to work with this hardware and AUTOSAR MCAL drivers. Looking forward to hearing from anyone with experience or advice

poula10, 2024-12-18:
Hi @marcmo and @christian-schilling,

Could you please help us with this problem? Your expertise would be greatly appreciated. Let us know if you need more details or context.

Thank you!

mthiede-acn2, 2024-12-20:
Hi all, great to see your interest in OpenBSW!

Regarding the original questions above:
1. At the moment OpenBSW does not support the PowerPC architecture
2. Multicore support is on our roadmap. The first version we intend to provide will use separate OS instances per core and we plan to provide inter-core communication mechanisms via shared memory
3. There is no general problem using C drivers from the C++ stack. Of course, when using OpenBSW, you will need to provide the C++ environment it relies on. This means for example certain changes to the linkerscript and linking some C++ libs, you can see this in our "referenceApp" example.

We do not currently support putting OpenBSW onto an AUTOSAR MCAL. Nevertheless, the current core stack is relatively easy to port as it has only few dependencies to the underlying driver layer. 

One of the first steps would be to have FreeRTOS running on the PowerPC board. Alternatively, our "async" OS abstraction layer could be adapted to another OS which might be running already. This is a little more involved, though. In the latter case you would have to provide your own asyncYourOs module, as a replacement to the existing asyncFreeRtos module which adapts "async" to FreeRtos.

On top of that, you would need to provide a CAN adapter (an implementation of can::AbstractCANTransceiver, instead of the existing bios::CanFlex2Transceiver), system time functions (see SystemTimer.h), serial IO functions (getByteFromStdin(), putByteToStdout()) and interrupt lock/unlock functions (see SuspendResumeAllInterrupts.h).

We are working on ports of OpenBSW to Zephyr and ThreadX. Once these are out, it will become more clear how OpenBSW can be ported. In the meantime, we hope to provide more information about porting soon.