eclipse-openbsw/openbsw #86: Middleware / RPC support

rolandreichweinbmw, 2025-04-02:
We are planning to contribute a Middleware implementation in terms of RPC interface for events and method calls across cores when OpenBSW is instantiated on multiple cores.

Communication will be done via shared memory.

vladyslavmarkovaccenture, 2025-06-17:
@rolandreichweinbmw since we're planning to switch to ETL, you can use ETL.

rolandreichweinbmw, 2025-07-10:
I could only open the same kind of issue again. The only option I currently have is to call it either "Bug Report", "Feature Request", "Blank Issue", "Security Vulnerability". But I don't have the credentials to add the "enhancement" label.

(Unfortunately, I also can't re-open this issue by myself, so please decide if you re-open the issue or we set up a new one.)

mthiede-acn2, 2025-07-10:
Reopening this issue to represent the middleware enhancement.

mthiede-acn2, 2025-07-10:
It would be great if you could briefly describe the planned feature here.

rolandreichweinbmw, 2025-07-10:
(Edited the initial description above.)