eclipse-openbsw/openbsw #88: check if gcc version in code coverage can be kept in sync with our build action version

marcmo, 2025-04-03:
matching gcov version would have to be used

shamitha-shashidhara, 2025-04-04:
Hello @marcmo ,
The GCC version we are using in code coverage and GCOV matches the version mentioned in the OpenBSW documentation. 
In the build.yml, we are not explicitly installing GCC, instead, we are using the GCC provided by the ubuntu-latest VM. To keep both in sync, I recommend using GCC version 11 in build.yml instead of changing the code-coverage.yml. 
Since we have not explicitly installed GCC in build.yml, the GCC version changes depending on the Ubuntu VM update on GitHub.

shamitha-shashidhara, 2025-04-11:
I have made the necessary changes in the build action to keep it in sync with the code coverage gcc version. Here is the PR: https://github.com/eclipse-openbsw/openbsw/pull/95

closed 2025-07-02 by marcmo