eclipse-openbsw/openbsw #90: The new cache usage in CI breaks build on removed files

rolandreichweinbmw, 2025-04-05:
The recently introduced cache usage in CI breaks builds if files are removed in a PR.

See e.g. https://github.com/eclipse-openbsw/openbsw/actions/runs/14263565171/job/39980303173

As you can see in this PR, iterator.cpp is (intentionally) removed, but the CI generates an error by referring to old cache contents which causes a "cannot read" error on this file.

By reverting the commit 8bbe6f0 ("CI: Integrate sscache for faster builds"), I can confirm that the issue is caused by this change from 8bbe6f0. I switched it back and forth in https://github.com/eclipse-openbsw/openbsw/pull/58 by force pushing for demonstrational purposes:

https://github.com/eclipse-openbsw/openbsw/actions/runs/14280714267/job/40029919670
https://github.com/eclipse-openbsw/openbsw/actions/runs/14280777692/job/40030055738

@shamitha-shashidhara and @marcmo , please find a better cache strategy or revert 8bbe6f0.

Thanks!

shamitha-shashidhara, 2025-04-07:
Hi @rolandreichweinbmw and @marcmo,
We were using the `restore-keys` option with actions/cache@v3, which restores a cache that is close to the current key but not an exact match. This was keeping a copy of the previous PR cache in the cmake-build-unit-tests directory, leading to a "cannot read" error. I am removing the restore key in the fixed PR. Also, to be on the safer side, I am deleting the cmake-build-unit-tests directory before running the unit test build. Additionally, I have added --ignore-errors source to make sure the HTML generation does not fail if the file is not found. Here is the PR [https://github.com/eclipse-openbsw/openbsw/pull/91] that fixes the above error.

rolandreichweinbmw, 2025-04-07:
The issue does not occur anymore in https://github.com/eclipse-openbsw/openbsw/pull/58 - so I'm closing this issue.

Thank you @shamitha-shashidhara for the fast fix.

closed 2025-04-07 by rolandreichweinbmw