eclipse-openbsw/openbsw #162: StderrLogger.cpp linking to libc instead of libs/3rdparty/printf

jiangyilism, 2025-08-26:
In file libs/bsw/util/src/util/logger/StderrLogger.cpp , 
StderrLogger::logOutput() is calling fprintf() , which links to libc instead of libs/3rdparty/printf . 
 
Should it link to libs/3rdparty/printf instead?

johannes-esr, 2025-08-27:
@jiangyilism, thanks for pointing out, I'll have a look with my colleagues!

vladyslavmarkovaccenture, 2025-08-27:
StderrLogger.cpp doesn't use 3rdparty's printf, because it doesn't call any of it's methods (fprintf / vfprintf - are not printf's methods). I also don't see any mentions of StderrLogger in the map file, meaning that it's probably opt-out. In other cases 3rdparty's printf is used (e.g vsnprintf_), but in other libraries.